Voxel8/Marlin #17: Don't shut down immediately upon thermistor disconnect

jminardi, 2015-07-10:
It might be nice to not shut down immediately upon thermistor disconnect. Sometimes the cartridge is bumped and the thermistor is not read for a short time. This immediately causes Marlin to shut down, which means we can't move the axis.

It might make sense to disable all heaters when the thermistor is disconnected, but still allow the axis to move.

kdumontnu, 2015-08-13:
Update - Default 4s buffer is added before Marlin will pass an M999 error. 

We should handle this better in the future. We will want to stop motors when any thermistor (bed or cartridge) is disconnected, as it is likely that we have crashed...

riningerr, 2015-10-13:
Update - grace period is currently 100 ms for hotend, 1 s for bed.

danthompson41, 2016-03-21:
See Cartridge.cpp, which now handles cartridge removal / thermistor disconnect

closed 2016-03-21 by danthompson41