Voxel8/Marlin #20: more to try with software-induced stepper motor skipping

Mikenanotube, 2015-07-20:
In Voxel8/DevKit-Issues#7 In the Integration branch of Marlin, for certain motors, certain speeds, and certain amounts of microstepping, skipping starts happening during motor movements, even when torque appears to be high enough to make the movement properly, and the speed, acceleration and Jerk are relatively low.

We need to find the code in Marlin that defines the signal sent to the stepper motor drivers, compare that code and related code between the gen3 dev and integration branches, and add print statements so that we can see the signals while Marlin is moving the motors, to aid debugging. We should also research microstepping to better diagnose what is happening

There is one hypothesis we haven't tested yet: it seems there is a safety factor built into the digipot stepper motor currents (0 to 255 scales from 0 to 1.4 amps, not 0 to 2 amps?), according to some old documentation from Steve Kelly. if this safety factor has been further reduced, that would futher limit the max torque

the issue we are having seems to be more an intermittent failure to send the proper signals to the motor drivers though, so we need to go into marlin and debug this to fix it.

Yes there are speeds where we can print without skipping, but we should be able to print many times as fast as what we're doing without worrying about skipping

Mikenanotube, 2015-07-20:
another thing I just noticed is that we originally forked from Steve Kelly's Marlin, not from the main branch of Marlin - we should check the difference between main marlin and Steve's marlin to see what he did differently with motor driving.

kdumontnu, 2015-08-13:
Will add this to a larger task I have, which includes determining optimal current limit/motor selection.

kdumontnu, 2015-10-12:
A patch of Marlin fixes this issue:
https://github.com/MarlinFirmware/MarlinDev/issues/35#issuecomment-136103737

closed 2015-10-12 by kdumontnu