Voxel8/Marlin #85: Fix Warnings from build script

danthompson41, 2016-02-17:
There are a bunch of issues when you build the code. We should take care of them.

kevingelion, 2016-02-17:
can you elaborate on the issues? based on what version of the Arduino IDE you have installed, warnings get more prominent - the build script was set to only be used with Arduino 1.6.1 i think.

danthompson41, 2016-02-17:
Sure. I just installed 1.6.6 which must be the cause.  The output is as follows:

>Voxel8 Marlin Build Script v1.0
>
>Loading configuration...
>Initializing packages...
>Preparing boards...
>Verifying...
>sketch\Marlin_main.cpp:262:43: warning: narrowing conversion of '0.0' from 'double' to 'uint16_t >{aka unsigned int}' inside { } [-Wnarrowing]
> static uint16_t bedlevelprobes[9] = { 0.0 };
>                                           ^
>In file included from sketch\Marlin.h:20:0,
>                 from sketch\Marlin_main.cpp:30:
>sketch\Marlin_main.cpp: In function 'void setup()':
>sketch\_Version.h:10:32: warning: unknown escape sequence: '\/' [enabled by default]
> #define DETAILED_BUILD_VERSION "dt\/refactor-remove_sd_support-43b3c8f-modified"
>                                ^
>sketch\Marlin.h:91:29: note: in expansion of macro 'SERIAL_PROTOCOLLNPGM'
> #define SERIAL_ECHOLNPGM(x) SERIAL_PROTOCOLLNPGM(x)
>                             ^
>sketch\Marlin_main.cpp:738:3: note: in expansion of macro 'SERIAL_ECHOLNPGM'
>   SERIAL_ECHOLNPGM(DETAILED_BUILD_VERSION);
>   ^
>sketch\Marlin_main.cpp:738:20: note: in expansion of macro 'DETAILED_BUILD_VERSION'
>   SERIAL_ECHOLNPGM(DETAILED_BUILD_VERSION);
>                    ^
>sketch\Marlin_main.cpp: In function 'void gcode_M115()':
>sketch\language.h:127:258: warning: unknown escape sequence: '\/' [enabled by default]
> #define MSG_M115_REPORT                     "FIRMWARE_NAME:Marlin " DETAILED_BUILD_VERSION " >SOURCE_CODE_URL:" SOURCE_CODE_URL " PROTOCOL_VERSION:" PROTOCOL_VERSION " MACHINE_TYPE:" >MACHINE_NAME " EXTRUDER_COUNT:" STRINGIFY(EXTRUDERS) " UUID:" MACHINE_UUID "\n"
>                                                                                                                                                                                                                                                                  ^
>sketch\Marlin_main.cpp:4292:3: note: in expansion of macro 'SERIAL_PROTOCOLPGM'
>   SERIAL_PROTOCOLPGM(MSG_M115_REPORT);
>   ^
>sketch\Marlin_main.cpp:4292:22: note: in expansion of macro 'MSG_M115_REPORT'
>   SERIAL_PROTOCOLPGM(MSG_M115_REPORT);
>                      ^
>sketch\Marlin_main.cpp: In function 'void gcode_M241(long int)':
>sketch\Marlin_main.cpp:5203:27: warning: NULL used in arithmetic [-Wpointer-arith]
>   if (num_milliseconds == NULL) {
>                           ^
>In file included from sketch\Configuration.h:5:0,
>                 from sketch\Marlin.h:26,
>                 from sketch\temperature.cpp:21:
>sketch\temperature.cpp: In function 'void tp_init()':
>sketch\macros.h:5:22: warning: left shift count is negative [enabled by default]
> #define BIT(b) (1<<(b))
>                      ^
>sketch\temperature.cpp:1089:82: note: in expansion of macro 'BIT'
>     #define ANALOG_SELECT(pin) do{ if (pin < 8) DIDR0 |= BIT(pin); else DIDR2 |= BIT(pin - 8); >}while(0)
>                                                                                  ^
>sketch\temperature.cpp:1101:5: note: in expansion of macro 'ANALOG_SELECT'
>     ANALOG_SELECT(TEMP_0_PIN);
>     ^
>sketch\macros.h:5:22: warning: left shift count is negative [enabled by default]
> #define BIT(b) (1<<(b))
>                      ^
>sketch\temperature.cpp:1089:82: note: in expansion of macro 'BIT'
>     #define ANALOG_SELECT(pin) do{ if (pin < 8) DIDR0 |= BIT(pin); else DIDR2 |= BIT(pin - 8); >}while(0)
>                                                                                  ^
>sketch\temperature.cpp:1113:5: note: in expansion of macro 'ANALOG_SELECT'
>     ANALOG_SELECT(TEMP_BED_PIN);
>     ^
>sketch\macros.h:5:22: warning: left shift count is negative [enabled by default]
> #define BIT(b) (1<<(b))
>                      ^
>sketch\temperature.cpp:1089:82: note: in expansion of macro 'BIT'
>     #define ANALOG_SELECT(pin) do{ if (pin < 8) DIDR0 |= BIT(pin); else DIDR2 |= BIT(pin - 8); >}while(0)
>                                                                                  ^
>sketch\temperature.cpp:1116:5: note: in expansion of macro 'ANALOG_SELECT'
>     ANALOG_SELECT(PNEUMATIC_PUMP_PIN);
>     ^
>Bootloader file specified but missing: C:\Program Files (>x86)\Arduino\hardware\arduino\avr\bootloaders\stk500boot_v2_mega2560.hex
>
>Sketch uses 84,270 bytes (32%) of program storage space. Maximum is 258,048 bytes.
>Global variables use 7,182 bytes of dynamic memory.
>The action has been performed successfully.

What's the best way to approach this? I can move to 1.6.5 if that's the stable version we're using

kevingelion, 2016-02-17:
Yeah the best way is to move to 1.6.5. They made big changes in recent versions that I haven't had time to account for.

danthompson41, 2016-02-17:
Perfect, I'll go to 1.6.5. Thanks!

danthompson41, 2016-02-17:
No warnings on 1.6.5, I'll close the issue. Thanks Kevin.

closed 2016-02-17 by danthompson41