Voxel8/Marlin #91: Need Mcode to modify steps per unit

danthompson41, 2016-03-14:
We want to be able to modify steps per unit, based on micron / mm. Carmen should know the details of what we want to see from it

sean-v8, 2016-03-15:
@danthompson41, does `M92` already take care of this?

danthompson41, 2016-03-15:
I actually saw that, and figured it might. We may want to modify it / duplicate it to take in Micron / MM, as that's what our measuring equipment spits out. I believe it does cover the same ground though

lewisgroup, 2016-03-16:
This should be lower priority. I brought it up yesterday because I thought
it would fix an immediate problem Carmen had, but it turns out it will not.
That being said this could still be useful and should be implemented if
there is time to spare.

danthompson41, 2016-04-21:
M272!

closed 2016-04-21 by danthompson41