Voxel8/Marlin #167: Can't flash the firmware to the Voxel8 printer with Arduino?

MichaelAPerrone, 2018-03-06:
_version.h appears to be missing so automatic versioning in configuration.h has to be commented out, then for some reason 3 extruders are enabled instead of 2 so sanitycheck.h fails. (maybe the laser distance sensor is counted as an extruder?)

After those, various pins appear to not be defined and it's a bit of a game of whack-a-mole. I think Arduino needs more information about the electronics hardware to successfully compile. Does flashing work with some alternative to Arduino?

MichaelAPerrone, 2018-03-06:
I think I see: Octoprint is used instead of Arduino. It doesn't work with my system because there is some communication issue between the RAMBO and the raspberry pi. Probably an issue with the Rambo

danthompson41, 2018-03-06:
Sounds about right. I would try to figure out where _version.h should be coming from, rather than commenting out features that butterfly effect :)

closed 2018-03-06 by MichaelAPerrone